rosmo-robot/Open-Core-M5stack #1: Review ESP32 adaptor board

samuk, 2022-01-04:
Does this seem like it would work?

runger1101001, 2022-01-04:
Another comment: I see that most of the m5 header pins are not connected. But if this is to be a replacement open core board for m5 stack, all the ESP32 pins should be connected to the corresponding m5 header pins, or it won't be compatible with the standard core and the other modules...

samuk, 2022-01-04:
Thanks, I uploaded it here without really checking it. I think we're missing the whole bottom layer. I'll get back to the guy and see if he can export again/resolve.

samuk, 2022-01-04:
If you open the .brd file in Fusion360 you can see the bottom layer. I can't get it very clear, but I don't know 360 really.

[image]

samuk, 2022-01-05:
"Did you cross-check the pins used against the ESP32's flash pins? It has a bunch of pins you can't really use, or only after boot-up is complete. It's usually a mistake to connect things to these pins, it can prevent the ESP32 booting."

esp32 flash pins are gpio 1 and 3 and those pins are connected with both m5 connector and uext connector

Would you recommend disconnecting them and choosing some alternates for GPIO?

runger1101001, 2022-01-07:
Hey, the new picture looks much better for the header connections.

Now I have the following points:
- I'd say its usual for nCS lines to have pull-ups...
- I'm still quite confused around the M5 power concept. In this configuration, the MCU boards 5V, wherever it comes from, would power the stack, or the stack would power the 5V. But the 3.3V is also connected, so the MCU board's 3.3V regulator will power that. What if another module applies power to this pin? These topics are pretty hairy, you can take a look at some of the recent Arduino MCUs to see how they solve it with their VIN, 3.3V and 5V rails... they have a bunch of mosfets (as ideal diodes) and transistors to sense/switch the power paths. Is it necessary? I don't know. Depends how the different components involved handle being back-powered...
- VBAT is unconnected, so the open-core won't support battery power/charging like the normal core does...

Flash pins: are GPIOs 6,7,8,9,10,11 - these pins should not be used, really, but it looks like they aren't exposed on the MCU board anyway, so all good there. Pins 1,3 are the default UART, you can use those.

samuk, 2022-01-07:
Comments from developer
"you don't have to worry about those 3.3v and vin/5v.

as long as you are using 3.3v in that 3.3v pin it will be fine (input or output). as esp32 dev board already have a voltage regulator inside. But you have to supply 5 v or more in the vin pin .

I checked those arduino mcu vin 3.3v and 5 v rail. those are same as esp32 dev board one.

and you will not need any mosfet or diode for sense or switch. (i tested that myself in my some other pcb) . you will need that if you want to keep esp32 powered by battery and also charge the battey using vin."

ShouravSarkar, 2022-01-07:
Hi there. 
vbat pin is exposed because there is not connection pin for battery from esp32 dev board. we can still use vbat but we have to make separate circuit for that which will power the board and also charge the battery from the VIN supply. 

there are some board like adafruit esp32 board which is open source  and they break out the vbat pin

runger1101001, 2022-01-07:
Much as I like the Adafruit products, I think that would make the whole thing more expensive... and specific to a vendor/board. As far as I know there are quite a few of the DevKit type boards out there, presumably most with compatible pinouts?
So I think that is actually a good choice.

I'll trust you on the power concept ;-) if you say you've tried it out in the past, that's what's most important.

Maybe a solution for the VBAT is to use an existing battery-charger/boost converter board - there are many on aliexpress... or make the decision to keep it as it is, and open-core just doesn't do the battery charging...
For Rosmo, there needs to be a different charging concept for its 3 LiFe batteries anyway...

ShouravSarkar, 2022-01-07:
can you tell me one thing? you are going to power the board using vin . right? and 3.3v will be output for other sensors that use 3.3 v. right? 

for battery charging you can use those  aliexpress module. but it is better to integrate so single vin will power everything

runger1101001, 2022-01-07:
Hey, when you say "the board", you mean the DevKit MCU board? I guess that's up to you, depending on your design for the open core board...

AFAICT the power concept for M5 stack seems to be mainly the 5V line, which either powers the core or is powered by the core. There's also the VBAT pin, which is expected to be powered by a Lion/LiPo 3.7V 1S cell. This pin either powers the m5 core, and via it the 5V line of the m5 stack, or it is powered by the m5 core's built-in charger, when the m5 core is plugged into USB.
VUSB isn't exposed unfortunately, so you can't charge the battery from another source.

The 3.3V line can be used to power 3.3V peripherals/modules, we'll use it to power the prototype ESC board when not connected to USB.

ShouravSarkar, 2022-01-08:
yes.

as your description, this PCB will be okay. 

only thing is the battery now. if you want to use battery then we have to use another battery charging circuit and also some mosfet diode etc

samuk, 2022-01-08:
I think the board as it stands works for our use case? Just to re-state to check my understanding:

- ~12v will be provided to the ESC to drive the motors
- An external 12v>5v buck will provide 5v to the ESC USB. This will power the M5stack Mbus 5v line.
- The Open core board discussed here will connect to the 5v-M5stack-Mbus-pin, which will power the board.

Is that correct?

For use cases beyond this, where people want to use a 3.7v Lipo, can they not use the [ESP32-Olimex-Lipo-board](https://www.olimex.com/Products/IoT/ESP32/ESP32-DevKit-LiPo/resources/ESP32-DevKit-Lipo-GPIOs.png) and connect the Lipo directly to the ESP32-Olimex-Lipo-board? So long as they then apply 5v to the 5v-M5stack-Mbus-pin for charging, the ESP32 on-board circuit should handle the charging?

ShouravSarkar, 2022-01-08:
For use cases beyond this, where people want to use a 3.7v Lipo, can they not use the ESP32-Olimex-Lipo-board and connect the Lipo directly to the ESP32-Olimex-Lipo-board?

yes. If they connect battery with esp32 board it will work perfectly and it will charge when 5v apply to m5 stack

samuk, 2022-01-11:
This issue has served its purpose, check other issues in the queue.

closed 2022-01-11 by samuk