rosmo-robot/Open-Core-M5stack #2: Create BOM for Open core

samuk, 2022-01-08:
Header pins etc.

samuk, 2022-01-08:
@ShouravSarkar could you provide a BOM? I had a look at it by loading the .brd file in Kicad, but couldn't work it out.

ShouravSarkar, 2022-01-08:
I already provided I think. okay I can provide again.

runger1101001, 2022-01-09:
IIRC you only have 2 resistors on there, plus the headers. And the m5 bus header is only going in one direction.

How about replacing the SMD resistors with small through-hole models, then the whole board will use only THT components?
Then I would produce this board without any assembly for $5 at PCBWay, or $2 at JLCPCB... Hand-soldering those few through hole components won't be hard...

ShouravSarkar, 2022-01-12:
okay I can change that smd with through hole

ShouravSarkar, 2022-01-12:
should I do that ?

samuk, 2022-01-12:
Yes please!

 These are part number for the m5 header https://store.open-electronics.org/2x15-Pin-Headers-Socket-2,54mm-Male-Connector-5330-M5CONN2X15M

ShouravSarkar, 2022-01-12:
ok you can use this header pin with current design

runger1101001, 2022-01-12:
Careful! That's a male header, and the core board needs a female one...
but it doesn't matter, the footprint supports both THT and SMD headers, and male/female have the same footprint, so no need to change the layout. It's just a BOM question.

samuk, 2022-01-12:
Sorry this is the female: https://www.futurashop.it/connettore-per-m5stack-2x15pin-femmina-5330-m5conn2x15f?search=M5CONN

Could you move the resistors to the centre of the board, underneath the ESP32. Please could you also add silkscreen with the part number/spec. The use case here is if it's your first ever attempt at soldering.

closed 2022-01-21 by samuk