rosmo-robot/Open-Core-M5stack #4: Investigate servo control from UEXT

samuk, 2022-01-09:
See if using the UEXT pins we could have [2x pwm](https://microcontrollerslab.com/esp32-servo-motor-web-server-arduino/) which users could jumper.

Use case would be the absolute lowest cost version of Rosmo driven just by Open Core and [ 2x micro servo](https://www.pololu.com/product/1053) instead of BLDC. These could fit in the existing BLDC socket with a 3d printed adaptor.

runger1101001, 2022-01-11:
In my mind, this mixes up different concepts:

- in m5 stack, there are cores and modules. the open-core is a core, but the servo and sonar stuff should be on a module.
- on the other hand if you want a minimal cost board for robots, then we could drop the m5 stack support (which complicates things and costs $$$) and make a specific board for that. But honestly I see this niche well served by a huge host of cheap boards available on AliExpress, see designs like Osmo, Zerobot etc...

Putting it all together runs the risk of producing some kind of franken-core, not properly compatible with m5 concepts, but also not ideal as a basic robot board

samuk, 2022-01-11:
I take your point.

It's driven by interest from @ottodiy. If we can support their use case well then they might help with the distribution/ development of the board. I guess I hope it could be a board that sees you through a robotics journey 

1) Get a Rosmo/ Otto wheeled moving about on continuous micro servo & Blockly
2) Invest a bit more & make a walking Otto
3) Invest a bit more and add an official M5 stack motor driver 
3) Invest a bit more Add a RosmoESC(s) to give better motor control

runger1101001, 2022-01-11:
Ah, I see! I'll make the observation that Ottos head looks a bit like an m5 stack in shape... :-)

closed 2022-01-27 by samuk