rosmo-robot/Open-Core-M5stack #5: Add additional grove socket?

samuk, 2022-01-09:
[image]

How much of a nightmare would it be to add an additional grove socket in that top left position? We don't need it for our use case, but might make the board more useful generally.

runger1101001, 2022-01-10:
I think the pull-ups could easily be moved to make space. They can go under the dev-kit board, for example.

Note that I just heard back from PCBWay that the Grove connectors are sold out.

samuk, 2022-01-10:
Great, let's see if @ShouravSarkar has any enthusiasm to do it.

Presumably, we could design it into the board, but leave it unpopulated in a first run?

ShouravSarkar, 2022-01-12:
ok can you tell me what to update in a single details?

samuk, 2022-01-12:
Had a change of plan on this one. Please include a https://www.sparkfun.com/qwiic  (see FAQ at the bottom for spec)